mjrice/wilson #7: TS model ods/wiki vs txt BOM out of sync

dantman, 2015-01-01:
I see a number of differences between the txt BOM [in git](https://github.com/mjrice/wilson/blob/master/doc/wilson-bom-ts.txt) and the .ods BOM and [wiki BOM](http://reprap.org/wiki/Wilson_TS_Bill_of_Materials) for the TS model.

Namely in the size of the horizontal and vertical 2020s and M8 x and y rods.

mjrice, 2015-01-01:
Ack, they got a little out of sync.  I've fixed them, let me know if you run into any other trouble or confusion.

dantman, 2015-01-01:
Ok, I can give you notes on the little, not immediately understandable, bits I run into as I inch toward trying to build a Wilson TS.

((I'm hoping to build myself my first 3D printer, after several months of saving up and getting parts, and decided to try making a Wilson TS instead of going with a Prusa i3 kit))

Want me to use this issue for the small bits or open new ones for them?

----

For example, my next little note would probably be this one.

When I was going over the stl files, trying to collect the list of parts I'd need printed by a third party, so I could estimate the costs, I was confused by `stl/belt-guide-sleeve-624.stl`. It isn't mentioned in the readme, wiki, or videos so I didn't know what it was for. I was confused for a little while trying to make sure that wasn't a part missting from the list. Then the commit message gave me the hint that it's an alt version of `belt-guide-624.stl` where on side snaps around the other.

dantman, 2015-01-01:
`stl/ts/extruder/` has three different extruder bodies: `extruder-body.stl`, `extruder-body-clampmountstyle.stl`, and `extruder-body-clampmountstyle-1.75mm.stl` for the original style extruder.

There's no documentation on the differences so it's not immediately obvious what one would pick / what situation determines which to use.

It's also not clear if `duct-xlong.stl` is the other version of the duct, for when using the E3D head, mentioned in [the video](http://youtu.be/z2bXSRLNdbQ).

mjrice, 2015-01-01:
I think I'll create a subfolder for stls that are still here but not usually needed, and then I'll move those other versions of things there so that the default parts are more easy to identify.  Thanks for the feedback!

mjrice, 2015-01-02:
Ok, I've done that.  I'll close this issue and if you see anything else just let me know / open a new issue.

closed 2015-01-02 by mjrice

iMattmax, 2015-03-25:
What kind of endstops are in the Wilson TS used?