mjrice/wilson #10: M8 replacement for M5 Rods

greenaar, 2015-04-01:
Could you release a modification to the TS model that allows for M8 rods instead of M5?  I'm about to start building one of your printers, and I'd rather do M8 for my Z axis threaded vs the M5 you have spec'd.

mjrice, 2015-04-01:
Sorry, I don't have models that would use 8mm threaded rods, and it would probably not be a simple change due to the increased size of the nuts being too close to the linear bearings for the X ends (I'm guessing).  Also, it seems like that would be a step in the wrong direction as far as accuracy and print quality.  I have been thinking of how to integrate 8mm lead screws, which is a different thing also, but it would require even more extensive changes to the design to accommodate the (even larger) copper nut used with those.

greenaar, 2015-04-01:
Understood.  It looks like I could potentially replace ..well, most of the x axis assembly with one from another model that does M8 threaded (though i may just go get some m5 rod -- is it not more wobbly for being so much thinner?)

closed 2019-12-01 by greenaar