mjrice/wilson #13: Endstop spaces in printed parts do not have room for non-soldered limit switches using disconnect connectors

dantman, 2015-05-14:
I have a RepRapElectro RAMBo which comes with switches and connectors to use for endstops.

Instead of the limit switches having short terminal ends and being soldered to, these switches are slightly longer and a "disconnect" type connector/plug is used to attach the wires to the limit switch. This makes them a bit longer and bulkier than other endstops.
[image]

However none of the endstop slots on the Wilson TS have room for these switches. None are deep enough for the plugs. And some of them are shallow enough that the switches themselves don't fit properly.

Here are some photos of the switches next to the endstop slots.
[image]
[image]
[image]

I'll probably find some workaround to get my printer working. However it may be worthwhile to modify the printed part designs to be deep enough to accommodate these kinds of switches. The connectors do seem like a nicer idea than using solder everywhere. And they are being sold along with a board made for RepRap printers.

The Z-axis should theoretically be the easiest fix. It can simply be extended upwards a bit so there's room.

The Y-axis could be a bit trickier. A simple option would probably be to cut out a hole behind the endstop slot, that would allow the wires to simply exit from there. However it wouldn't look that nice and I'm not sure what affect if would have on the printability.

The X-axis will be the most problematic. With the threaded rod in the way there's basically no room for the endstop slot to go any deeper. And if you move it to the other side of the screw holes the switch doesn't hit the carriage.

dantman, 2015-06-05:
The x-axis is actually even worse than I thought. I bought a set of smaller switches. The z and y axis are fine with those. But the x-axis actually doesn't even fit a smaller switch ([these](http://www.robotdigg.com/product/141/Endstop,-Snap-Action-Limit-Switch-SS-5GL)).

The switch holder on the x-axis should be extended a little. But I'm starting to think that the x-axis should be redesigned so that the endstop switch is located on the back of the motor end and the back of the carriage has an extra bit of plastic that will hit it. (Though I personally have some other redesign ideas I want to try on the x-axis anyways)

mjrice, 2015-06-05:
I'm not sure why you wouldn't just solder the wires to the endstop, and then it would fit as-is.  Do you need disconnect capability on the switch?

dantman, 2015-06-05:
Technically no. But personally I like how elegant it is to simply plug in instead of messing with a soldering iron and I loath the idea of butchering the endstop cable that came with my RAMBo. Practically speaking, since there are switches sold like this in a product made for 3d printers I think it would be reasonable to support them.

Also keep in mind it's a little more involved than you think. Remember that the endstop slots on most of the axis are actually so shallow that the switches themselves sans-connector don't even fit in the slot. I'd have to take a nice switch+connector set, cut off the connectors, cut off 2/3 of the metal from the terminals on the switch, and then solder wires directly onto the switch directly onto metal (since the holes would no longer be present).

mjrice, 2015-06-06:
I'm not sure how long the terminals on your endstops are, but mine fit fine.  The switches I use are about 14mm from the lever to the end of the wiring posts, and the X axis pocket allows for 15mm.  I don't disagree with you that it's better to be compatible with as many switches as possible, but I don't fancy having to use longer rods or extrusions just to accomplish it. Mounting to the back of the X motor end isn't a bad idea either, but it would require some careful mods to the X carriage to ensure it worked consistently.  But I'd love to see what you come up with, improvements are always welcome!

dantman, 2015-06-06:
The most important measurement for the switches is probably the distance between the centre of the screw holes to the tip of the wiring posts.

The switches that came with [the RAMBo](https://reprapelectro.com/product/rambo/) measure:
- End of lever to end of wiring posts: 25mm
- Hole centre to wiring post tip: 11mm
- Wiring post length: 8.5mm
[image]

The ones I bought from RobotDigg measure:
- End of lever to end of wiring posts: 19.5mm
- Hole centre to wiring post tip: 6mm
- Wiring post length: 4mm
[image]

These fit into the y and z axis fine. But have a problem with the x-axis.

The problem there with the RobotDigg switches appears to be that the printed parts have 4-5mm between the screw holes and a wall. There is a gap with additional space but the terminals on these are 4mm perpendicular to the switch (which itself is 6mm and the terminals are in the centre). This means that the terminals do not fit into the opening.

This is my current workaround to that:
[image]

----

I wanted to do some modifications to the X carriage anyways. Some things I hoped to experiment with were:
- Seeing if I could apply the technique [this belt holder](https://www.youmagine.com/designs/i3-adjustable-y-belt-holder-for-gt2-t2-5) and [this clamp](https://www.youmagine.com/designs/gt2-belt-clamps) use for attaching the GT2 belts. (No vertical shifting of the belt and as a bonus, no zip ties)
- Separating the lower linear bearing holder slot into 2 holders and eliminating the plastic support in between. To see if that provides enough flex to compensate for variances in cheap linear bearings (I'm currently using one bearing in each slot instead of 3 because putting 2 in one slot basically caused my bearings to  lock up #11).
- I *might* also experiment with some of the quick change design ideas others have had and see if I can make something that works nice on the Wilson.

I was also thinking of replacing the idler tensioner with something based on [this belt connector](https://www.thingiverse.com/thing:793499)'s tensioner design. But I might hold off on that and see how well the double idler tensioner you have now works out.