mjrice/wilson #15: Sorry - not an issue.  Question.   What should I consider when.....

Snarfer, 2015-06-12:
Thinking about building this out with a 300mm square bed, bowden extruder, maybe 400mm high?  Would like to hear your thoughts on potential difficulties we might encounter on stretching this out.   Using Marlin firmware.  A Bowden recommendation would be appreciated.    I have always appreciated your thoughtful innovations to the Prusa - my son and I built an i3 about 3 years ago.    What head are you using?   We are considering an E3D over our original JHead.

Nice work, btw - hope your kits sell well!   I might consider buying one if I could select some mods ;)

Cheers!

mjrice, 2015-06-12:
I think that if you stretch the X axis that much, you may have more deflection in the X axis rods than you'd like.  The Y axis doesn't experience that much since the bearings are much farther apart (and it doesn't have the weight of the extuder cantilevered out like the X axis does).  

As for the Z axis to 400mm, I think you'll be ok as long as you use the angled side braces and mount the machine to a wooden deck.  That will make the upright frame stiff enough that the extra height shouldn't be a problem.

I am using both the original JHeads and the e3d-lite6.  I like them both.  Hope that helps!

Snarfer, 2015-06-13:
Great point on the x-axis.   How about a 3rd rail on the x-axis?   I
thought first to just make the two thicker - but the momentum around the
current rods would make sourcing them easier.   Tweak the carriage for the
hot end to add a third rail so you could stretch it out to 400, 500?
Haha!  Stiffen up that expansion and your off and running!