mjrice/wilson #16: Z-feedrate max for the Wilson

dantman, 2015-06-13:
What is the maximum z-feedrate others have been able to handle on the Wilson?
((3rd number in `#define DEFAULT_MAX_FEEDRATE {...}` in Marlin))

My threaded rods gave horrible screeches when my z-axis travelled down when I first put it together (self sourced parts) and I had to limit myself to 1 mm/s until the super lube I ordered arrived. With lubrication there's still a little metal-to-metal noise but some of it is gone.

My z stepper drivers are set to 1/4 microstepping and 1000 steps per unit. My z-feedrate max is currently set to 2 mm/s now. Setting it to 3-6 mm/s leads to my stepper motors locking.

mjrice, 2015-06-13:
Personally, I am running 2-3 mm/sec on different machines.  I have the microstepping set to 16, so 4000 steps per mm.

dantman, 2015-06-13:
IIRC with my stepper motors if I used 1/16 microstepping with 4000 steps I always ended up with my stepper motors locking up.
I wonder if there's a big difference between our stepper motors.
These are the ones I bought:
http://www.aliexpress.com/item/Free-Shipping-5pcs-4-lead-Nema17-Stepper-Motor-48mm-78Oz-in-1-8a-Nema-17-motor/1871797056.html

Or perhaps, tuning the z-axis to go faster is a matter of finding the right acceleration that gives the stepper motors time to spin up before they get to the higher speeds.

stephenci, 2015-07-13:
I have the kit and the z axis travel down screeches too. with default feedrate set at 5 per configuration.h the motor locked up. I have since change it to 2 and it works now but it also screeches. which part of the rod did you apply lubricant?

dantman, 2015-07-13:
I went a little overboard with my PTFE-fitted super-lube. Coating the whole thing (multiple times in some spots) as it was spinning and even using the gap in between the plastic that holds the nut and the rod as a kind of reservoir for lubricant, letting it coat the bar as it moved upwards.