mjrice/wilson #18: cable management

stephenci, 2015-06-30:
when I looked at lulzbot mini with the cable link carrier it just look so cool I wonder if anyone has attempted to incorporate the link carrier into wilson ts? any suggestion?

dantman, 2015-07-10:
Got a picture/link of what it looks like on the LulzBot Mini? I couldn't figure out what you're referring to.

There are a lot of different types of cable management pieces on thingiverse/youmagine. Most notable would be the ones designed to attach to 2020 extrusion.

stephenci, 2015-07-13:
here you go. picture 6/8 shows a good picture of the link carriage I was referring to.

http://www.pcmag.com/slideshow_viewer/0,3253,l=331798&a=331798&po=6,00.asp

dantman, 2015-07-13:
Oh a cable chain. There are a bunch of designs for those on Thingiverse. And pull-request #6 has a design for a Wilson mount for one.