mjrice/wilson #24: No SD Card

Gordon44, 2016-09-03:
Why do I get this message No SD Card, I have a good SD card in the slot so how do I fix this problem
the unit was working just fine and now I can't see the SD Card

timmit99, 2016-09-04:
Check your lcd cables. One of them is.dedicated to the SD card operations and if it becomes loose it won't register the card as inserted. Make sure they are pressed in firmly to the adaptor and the screen.

Gordon44, 2016-09-04:
Thank you that did it

Gordon44, 2016-10-25:
Hi Marty could you help me one more time 

I have been using my Wilson II 3D Printer now for several months now with no problems until now 

When I start the print job after I do the Bed leveling and Auto Home the printer starts to print but the print head is about 2 Inches from the bed 

I tried to print using some of my older files with the same results 

The print head is in the same location and has not moved 

The Z offset set is set to 11.5 this is where I have been setting now for months 

The Auto Home and the Bed leveling works just fine and all the switches test OK 

Then when I start the job the printer starts just fine however the print head is about 2 inches above the bed 

What should I do to fix this? 

Thank you 

Gordon Groves 

Gordon1@q.com 

----- Original Message -----
From: "Gordon Groves" <gordon1@q.com> 
To: "mjrice/wilson" <reply@reply.github.com> 
Sent: Saturday, September 3, 2016 7:48:09 PM 
Subject: Re: [mjrice/wilson] No SD Card (#24) 

Thank you that did it

rodelvalle, 2016-10-25:
I know this is going to sound strange but there is a possibility it is the motor drivers.  At least for me it was. After completing both the homing and auto-level the nozzle was printing on mid-air.  If it is the same issue, you can even move the nozzle manually close to the bed.  I used for my Wilson the recommended KL17H248-15-4A.  I found that when they are in parallel (z-axis) is to much for the A4988 Stepper Motor Drive, then freezing at that precise moment.  The solution was to install instead a DRV8825 and setting the current to almost 2.7A (1.35A for each motor).  I hope this helps...

mjrice, 2016-10-25:
> 11e6-95fc-7290892c7bb5.png ","avatar_image_url":" https://cloud.
> githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-
> 11e6-9aed-b52498112777.png ","action":{"name":"Open in GitHub","url":"

Hi Gordon,
When it is moving to home, does the LCD screen say "hit" next to the "Z"?
Does the endstop actually touch the bed when it is supposed to?

Gordon44, 2016-10-26:
When I run the program it will go to home and the print head will go down and touch the bed and beep and them say Z HIT . 

After that the Z axes will move up to Z = 20 and that is where the printer will start to print from in mid air . 

I have tried running other files that worked just fine it the past . 

So what is going on , why will the Z axis move up to 20 and then start printing ? 

Is the software bad if so should I try to reinstall it , if so can you give me the web address to down load the newest version , 
or is there something else going on that I should try . 

----- Original Message -----
From : "Gordon Groves " <gordon1@q.co m> 
To : "mjrice/wilson" <reply@reply.github.com> 
Sent: Tuesday, October 25, 2016 6:58:20 PM 
Subject: Re: [mjrice/wilson] No SD Card (#24) 

Yes it does say Z Axes Hit 
and yes the endstop touches the bed and beeps

Gordon44, 2016-10-26:
Yes it does say Z Axes Hit 
and yes the endstop touches the bed and beeps

Gordon44, 2016-10-27:
Good news I dissemble everything and re-assembled including the Arduino and now the printer works must have been a bad connection not sure just what? 

Could you email me the link to the Marlin program this website is a little confusing to just which program I should use I am planning on having a spare Arduino Mega 2560 just in case mine might go bad in the future? 

Oh also do you have any plans to build a Dual Extrude head version in the near future? 

Thanks for your help 

Gordon 

Gordon1@q.com 

505 281-4495 ----- Original Message -----
From: "Gordon Groves" <gordon1@q.com> 
To: "mjrice/wilson" <reply@reply.github.com> 
Sent: Tuesday, October 25, 2016 7:20:59 PM 
Subject: Re: [mjrice/wilson] No SD Card (#24) 

When I run the program it will go to home and the print head will go down and touch the bed and beep and them say Z HIT. 

After that the Z axes will move up to Z=20 and that is where the printer will start to print from in mid air. 

I have tried running other files that worked just fine it the past. 

So what is going on, why will the Z axis move up to 20 and then start printing? 

Is the software bad if so should I try to reinstall it, if so can you give me the web address to down load the newest version, 
or is there something else going on that I should try. 

----- Original Message -----
From: "Gordon Groves" < gordon1@q.com > 
To: "mjrice/wilson" < reply@reply.github.com > 
Sent: Tuesday, October 25 , 2016 6:58:20 PM 
Subject: Re: [mjrice/wilson] No SD Card (#24) 

Yes it does say Z Axes Hit 
and yes the endstop touches the bed and beeps