mjrice/wilson #25: Strain relief?

Xalcker, 2020-05-29:
I cannot find where is this piece needed in the assembly. Can anyone help? Is there a comunity or forma for this printer?

timmit99, 2020-05-29:
there was a forum but it seems to have lapsed and not been renewed. If I am looking at the part correctly the strain relief part is zip tied around the extruder motor and you can zip tie the extruder/toolhead wires to it. Ignore my crude MSPaint skills but this is how i envision it being used.
[image]

haroldj, 2020-05-29:
That piece is not really needed. I built the wilson rs about 4 years ago and it is still my go to machine and it still works great. The only thing different on mine is that I replaced the original extruder with a MK8 extruder. I have 2 other working printers, but this is still the one that gives me the best prints consistently.

timmit99, 2020-05-29:
I actually just tore apart my Wilson II after nearly 5 years of service. With my Voron 2 and Voron 0 i just didn't need the Wilson II anymore and i was having many issues with bed leveling. I'm keeping some parts as mementos though.

haroldj, 2020-05-29:
I have a wilson 2 that I built but never used because I couldn't get the config files any more after I finished the build. I have the original design with the rack and pinion levelling switch. Would you possibly have a copy of the config files? I am not good at editing arduino config files.

timmit99, 2020-05-29:
@haroldj i can LOOK but i switched to klipper about a year ago.

timmit99, 2020-05-29:
@haroldj wow, that was actually fast. Found my files from 11/7/16. FOR REFERENCE ONLY.  at this point i had a IR bed probe instead of the rack and pinion as i built mine before that was released.

[Configuration.txt](https://github.com/mjrice/wilson/files/4702865/Configuration.txt)
[Configuration_adv.txt](https://github.com/mjrice/wilson/files/4702866/Configuration_adv.txt)

haroldj, 2020-05-29:
Thanks a lot, that will help me a lot to get that up and running.. I was thinking of adding a regular z stop and use it that way.

Xalcker, 2020-05-29:
> there was a forum but it seems to have lapsed and not been renewed. If I am looking at the part correctly the strain relief part is zip tied around the extruder motor and you can zip tie the [...]

Ok, thankyou very much. I have built a Wilson 1, and still working fine, printing some replacement parts right now.